DiamondLightSource/pythonSoftIOC #1: Upgrade to Anaconda Python 3.7 breaks

prichards-wmko, 2020-02-28:
On CentOS 7 systems using Anaconda Python 3.6 in /usr/local/anaconda, we have been successfully using the pythonIoc for some time now.   The Anaconda package is installed via its own installer, not by any yum/rpm installers.

Recently I built another system (CentOS 7.7) with the latest anaconda3-2019.10 installed in the same place, /usr/local/anaconda.  I created a new, clean build of pythonIoc.  However, invocation of pythonIoc fails to find libraries and core dumps:

$ ./pythonIoc
Fatal Python error: initfsencoding: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'
 
Current thread 0x00007f1a8468a740 (most recent call first):
Abort (core dumped)

We investigated this error and learned that the errors relate to how Python searching for the standard libraries.  Troubleshooting articles suggest setting PYTHONHOME as defined here: https://docs.python.org/3.7/using/cmdline.html#environment-variables 
 
setenv PYTHONHOME /usr/local/anaconda

This appears to solve the problem.

$ ./pythonIoc
Python 3.7.4 (default, Aug 13 2019, 20:39:44)
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>

It's not clear why this was not a problem in the past.  PYTHONHOME has been around since the Python 2 days and I've never had to set it before.

We suggest maybe using the CONFIG_SITE file to establish the base Python directory, then create PYTHONHOME in pythonIoc.in based on that.

Araneidae, 2020-04-23:
Somehow I missed notification of this issue.  Thank you for the report, I'll see what's needed to fix it.

Araneidae, 2020-04-23:
I've added setting `PYTHONHOME` in the build process in a recent push.  Can you let me know if this works for you?

Araneidae, 2020-04-24:
I can explain why this has never been a problem in the past (I had to do too much digging to discover `PYTHONHOME`, wish I'd spotted this ticket _first_!)  As far as I can tell, `PYTHONHOME` is only needed when the library path baked into the Python interpreter is not the true library path, and this only seems to arise when using a Virtual Environment.

Araneidae, 2021-03-05:
I believe this issue is now fixed, will mark it as closed.

closed 2021-03-05 by Araneidae