DiamondLightSource/pythonSoftIOC #2: pythonIoc using a conda environment

abdomit, 2020-04-23:
This is what I did to use a conda environment with pythonIoc:
1. modify `configure/CONFIG_SITE`:
```
PYTHON = /operation/common/miniconda3/envs/mbf/bin/python3.7
PYTHON_CONFIG = $(PYTHON)-config
```
2. activate the conda environment
3. install python-config:
```
pip install python-config
```
4. call `make`
The first strange thing here is that it doesn't work if the conda environment is not activated, which is not what I was expecting since the path to `python3.7` is given.
But OK, the conda environment has to be activated, let's continue!
5. start the IOC
Then I get the following error:
```
/mntdirect_host/_operation_ro/control/os/debian9/epics/components/pythonIoc-py37/bin/linux-x86_64/softIoc: error while loading shared libraries: libpython3.7m.so.1.0: cannot open shared object file: No such file or directory
```
After adding the folder `/operation/common/miniconda3/envs/mbf/lib` in `LD_LIBRARY_PATH` it's OK.
Activating or not the conda environment doesn't make any difference: the correct path to `python` is used.

So I was wondering if there is a way to link `softIoc` with the correct path of `libpython3.7m.so.1.0`?

Araneidae, 2020-04-23:
Hmm.  Well, I've just pushed a patch for setting `PYTHONHOME`, which may fix #1, but sounds like your problem is different.

Araneidae, 2020-04-23:
I wonder if the problem is the `python-config` install.

How about instead setting `PYTHON = /operation/common/miniconda3/envs/mbf/bin/python` (without the `3.7` suffix), and not installing `python-config`?  I just tried something like that here, and that seemed to be enough: there should already be a `python-config` in the existing `bin` directory.

abdomit, 2020-04-23:
Actually there is no `python-config` in the `bin` directory. There is a `python3.7m-config`, and two link towards it: `python3.7-config` `python3-config`.

But I realized that this it is very likely that this `python3.7m-config` was here since the beginning.

I'll will try again with a fresh conda environment, and without installing `python-config`

abdomit, 2020-04-23:
I did the same procedure as in the first message, skipping step 3.
It gives exactly the same result.

I added this step 3. because I was surprise not to find a `python-config`, but in fact `python3-config` existed.