DiamondLightSource/pythonSoftIOC #3: [question] will this package be published to PyPi?

onichandame, 2020-07-06:
The README and docs both require to manually install this package. I wonder if there is a plan to publish it to PyPi so that we can use the `pip` tool to manage this package.

If yes, please tell the schedule. If no, please tell the reasoning behind.

// This tool is awesome! Thx for your work!

Araneidae, 2020-07-07:
Unfortunately, as currently implemented, pythonIoc must be generated as a standalone application, linked with your local EPICS installation, so I don't think this integrates terribly well with PyPi as it stands.

However, we are already looking at this.  @thomascobb, this is relevant to our meeting next week!

onichandame, 2020-07-07:
If I recall correctly, [pyepics](https://github.com/pyepics/pyepics) can be installed through PyPi and also makes use of the local installation of EPICS(if present). I wonder if we can lend some ideas from them.

I am a pure noob on EPICS so please forgive me if I am missing some common sense here.

thomascobb, 2020-07-08:
Some work that @mdavidsaver did on [epicscorelibs](https://github.com/epics-base/epics-base/pull/76#issuecomment-651305547) may help us here. He already did the hard work of packaging EPICS base for all the major arches via CI, and added a dynamic registerRecordDeviceDriver. We could either copy his work and build binary packages via CI here, or maybe even repackage this to be called from ctypes rather than building anything? Not sure if that would work... We'll discuss next week.

jalalmostafa, 2020-08-19:
What about a docker container? The container can provide the same functionalities as a pip package. The image would be built with a specific epics version (from source) and a specific python version (most probably from source too).
The disadvantage of this is that the image will be larger than the pip package but still manageable.

thomascobb, 2021-03-05:
Working on this in #5

Araneidae, 2021-06-28:
This is now available in beta form at https://pypi.org/project/softioc/.

closed 2021-06-28 by Araneidae