DiamondLightSource/pythonSoftIOC #7: [beginner question] example/runtest not running

MichaS-git, 2021-04-08:
After building pythonSoftIoc I get the following output when trying to run runtest:

bam01:/soft/epics/inst-3.15.7/pythonIoc/example # ./runtest 
Traceback (most recent call last):
filename="../../../src/ioc/dbStatic/dbLexRoutines.c" line number=259
No such file or directory dbRead opening file ioc.db
  File "startup.py", line 10, in <module>
    softioc.devIocStats('TS-DI-TEST-01')
  File "/soft/epics/inst-3.15.7/pythonIoc/python/softioc/softioc.py", line 161, in devIocStats
    'IOCNAME=%s,name=' % ioc_name)
  File "/soft/epics/inst-3.15.7/pythonIoc/python/softioc/softioc.py", line 156, in dbLoadDatabase
    imports.dbLoadDatabase(database, path, substitutions)
  File "/soft/epics/inst-3.15.7/pythonIoc/python/softioc/imports.py", line 11, in expect_success
    assert status == 0, 'Expected success'
AssertionError: Expected success
bam01:/soft/epics/inst-3.15.7/pythonIoc/example # 

What am I doing wrong?

Araneidae, 2021-04-08:
Ah, my apologies.  This needs improvements to the instructions.  I assume that you've already commented out the `DEVIOCSTATS` line in `configure/RELEASE` (I really ought to suggest that in the README!) ... to run the tests you also need to comment out the failing line 10 in `startup.py`, as the tests are also trying to load devIocStats.

With luck that should be enough.

MichaS-git, 2021-04-08:
I left the DEVIOCSTATS undefined, like it is recommended in the RELEASE file:

DEVIOCSTATS =

Commenting out line 10 in startup.py brought the solution, thanks!

closed 2021-04-08 by MichaS-git