DiamondLightSource/pythonSoftIOC #8: python path for user library seems to be broken

beach23, 2021-06-17:
/usr/bin/ld: /usr/lib/libpython3.6m.a(faulthandler.o): undefined reference to symbol 'pthread_sigmask@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
/home/test/myepics/base/configure/RULES_BUILD:230: recipe for target 'softIoc' failed
make[2]: *** [softIoc] Error 1
make[2]: Leaving directory '/home/test/Desktop/pythonIoc/softIocApp/O.linux-x86_64'
/home/test/myepics/base/configure/RULES_ARCHS:58: recipe for target 'install.linux-x86_64' failed
make[1]: *** [install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/test/Desktop/pythonIoc/softIocApp'
/home/test/myepics/base/configure/RULES_DIRS:85: recipe for target 'softIocApp.install' failed
make: *** [softIocApp.install] Error 2

After following directions, I get to the point where the user makes the BUILD_DOCS=, and it spits out this error.
It cannot find the directory for python. I have python installed and checked for any version problems.

Araneidae, 2021-06-18:
This is a very strange error, I've never seen anything like it.  Suspect I need more context.  Also, I don't see the connection between the error message you've posted and your title, which refers to the Python path.  At the very least, I need to see the command that is being executed to trigger the first error message.

Which version are you trying to build here?

beach23, 2021-06-18:
I am using EPICS 7 on linux. Even if I make clean and try to remake it gives me this error. I got the latest version off the website.

Araneidae, 2021-06-21:
Think you may be better off waiting for pythonIoc to be available on PyPi.  Think we're nearly there now.

@thomascobb , can you make any sense of this error report?  Otherwise think we'll have to close it.

thomascobb, 2021-06-21:
Sorry, not sure what the problem is here. We're almost done with #5, just docs left to do now, hope to be done in a week or so as I will be talking about it at the EPICS collaboration meeting on the 7th July!

Araneidae, 2021-06-21:
I'm guessing that issue #4, which seems to have passed me by, will also be swept up in this.  Think once #5 is done I'll close both these issues as resolved by #5!

beach23, 2021-06-22:
Here is the error pertaining to the python path. This is what I get from doing a regular make clean && make command.
I am using Ubuntu. 

make[2]: *** No rule to make target '/usr/lib/libpython3.6m.a', needed by 'softIoc'.  Stop.
make[2]: Leaving directory '/home/test/Desktop/pythonIoc/softIocApp/O.linux-x86_64'
/home/test/myepics/base/configure/RULES_ARCHS:58: recipe for target 'install.linux-x86_64' failed
make[1]: *** [install.linux-x86_64] Error 2
make[1]: Leaving directory '/home/test/Desktop/pythonIoc/softIocApp'
/home/test/myepics/base/configure/RULES_DIRS:85: recipe for target 'softIocApp.install' failed
make: *** [softIocApp.install] Error 2

Hope this helps a little.

Thanks,
Brian

Araneidae, 2021-06-23:
Well, I guess either you don't have the Python development libraries installed or something is wrong with the process for computing the library path.  Unfortunately you're still not showing me the commands actually being executed by `make`, so I'm having to guess.  Can you give me the output from the following commands please (I'm assuming you have set `PYTHON = python` in `configure/CONFIG_SITE`):
```bash
python-config --ldflags
```
and in the Python interpreter:
```python
from distutils import sysconfig
print(sysconfig.get_config_var('LIBRARY'))
print(sysconfig.get_config_var('LIBDIR'))
```
To be honest, this is largely academic: there is about to be a new release of pythonIoc which will completely change the build process; for a preview try the [on-its-head](https://github.com/dls-controls/pythonIoc/tree/on-its-head) branch.

Araneidae, 2021-06-28:
Try using release [3.0b1](https://pypi.org/project/softioc/).  I'll close this report now, as is effectively resolved by #5.

closed 2021-06-28 by Araneidae