DiamondLightSource/pythonSoftIOC #16: Failed to follow "Read data from an IOC" section of tutorials

Hep-dog, 2021-07-06:
Hi all,
After creating a cothread IOC, I tried to read data in another shell. But I got the following error :
[image]

I also tried to use the epics.PV.get, and it was worked:
[image]

I don't know why, and will be very grateful for your help. Thansk a lot!

Araneidae, 2021-07-06:
Chances are you have a local firewall blocking ports 5064 and 5065.

Note that you do have to run the IOC process by running `softioc.interactive_ioc()` after creating the PVs before they are visible from outside.

thomascobb, 2021-07-07:
Another thing to try, if you remove the `from softioc import softioc` in your client, what happens?

thomascobb, 2021-07-07:
Also, what OS and architecture is the machine you're running on?

Araneidae, 2021-07-30:
Closing as unable to reproduce and no response from reporter.

closed 2021-07-30 by Araneidae