DiamondLightSource/pythonSoftIOC #26: Weird channel access problem within Python softioc process

Araneidae, 2021-08-11:
From [thread tech talk](https://epics.anl.gov/tech-talk/2021/msg01538.php).

> A simplified example of the problem:
>
> 1.    Import the pyepics and softioc libraries.
> 2.    Start a softioc with a channel (call it ‘prefix:long’ in my example).
> 3.    Start a thread that periodically updates the channel with a new value.
> 4.    Start another thread that tries to access this channel, either with epics.PV() or epics.caget().
> 5.    Channel access in thread 2 always times out, forever.
>
> Today I found a “solution” to this that does not make sense: if I start the softioc (the call to `softioc.iotInit()`) in the first thread, and not in the main process, everything works as expected.

Attached example code:
```python
import os
import time
import threading

os.environ['EPICS_CA_SERVER_PORT'] = '5100'
os.environ['EPICS_CA_ADDR_LIST'] = 'localhost:5100'
os.environ['EPICS_CA_AUTO_ADDR_LIST'] = 'NO'
 
try:
    import epicscorelibs.path.pyepics
except ImportError:
    pass

import epics
from softioc import softioc, builder

# Create a long record for this demo
long = builder.longOut('prefix:long', initial_value=1)
builder.LoadDatabase()
softioc.iocInit()

def thread1():
    while True:
        val = long.get()
        long.set(val + 1)
        print(f'thread1: {val}')
        time.sleep(1)

def thread2():
    pv = epics.PV('prefix:long')
    while True:
        val = pv.get()
        print(f'thread2: {val}')
        time.sleep(1)

t1 = threading.Thread(target=thread1, daemon=True)
t2 = threading.Thread(target=thread2, daemon=True)

t1.start()
t2.start()

time.sleep(10)
```
If `iocInit` is moved into `thread1()` the example works.

Araneidae, 2021-08-11:
[Response](https://epics.anl.gov/tech-talk/2021/msg01540.php) from @mdavidsaver:

> You might try moving the 'import epics' line after softioc.iocInit().
>
> fyi. libca has some special behavior when run within an IOC process
to facilitate access to local records.  As I recall, this imposes
a required initialization order.
>
> Also, libca uses a thread local variable to hold the client context
pointer.  It is possible that pyepics is leaving its context pointer
in place, which will likely cause problems during iocInit().

thomascobb, 2021-08-11:
Hi Paul,

> Import the pyepics and softioc libraries.

Something to be aware of, the default in pythonSoftIOC is to use cothread as its concurrency framework. It uses lightweight co-routines instead of OS threads. This means that you must be careful not to block the thread you call softioc.iocInit() in, otherwise the callbacks from caput and myrecord.set() will end up not being processed. Functions like softioc.interactive_ioc() will co-operatively block, allowing cothread to service callbacks in the background.

I can suggest a couple of things to try:

-    Either continue using cothread for concurrency, and use cothread.catools instead of pyepics for channel access. This will co-operatively block, allowing cothread to keep servicing the record callbacks
-    Or don't use cothread for concurrency. The only other supported concurrency framework at the moment is asyncio, but you could use that example to make a background thread dispatcher that did the same kind of thing as AsyncioDispatcher, but with a single background OS thread

Please could you put responses on the pythonSoftIOC issue that Michael created to track this.

Thanks,
Tom

Araneidae, 2021-09-14:
I will close this by the end of October if there are no further updates.