DiamondLightSource/pythonSoftIOC #27: What is a "Publishable IOC"?

Robatronic, 2021-08-11:
Hope this is the right place for this question.....

Saw a reference to "Publishable IOC" in the docs. Is that a Diamond-specific definition or EPICS v4 or something else? What is it?

Cheers,
Tanner

Araneidae, 2021-08-12:
I guess you're referring to [this page](https://dls-controls.github.io/pythonSoftIOC/master/how-to/make-publishable-ioc.html).

I agree, that does look very DLS internal specific.

Also, @thomascobb , it strikes me that the shebang direct reference to the pythonIoc executable is a bit unnecessary; I thought the whole point of the recent effort was to mostly avoid the standalone application?

thomascobb, 2021-08-12:
@Robatronic yes this is a DLS specific definition. I should add a note to say that.

@Araneidae the pythonSoftIOC executable is for backwards compatibility for modules in /dls_sw/.../ioc (as there's no dls-python3 equivalent with numpy, pythonSoftIOC, scipy, cothread, etc in). New IOCs are encouraged to be Python modules in /dls_sw/.../python3, which means they will manage the versions of their dependencies via pipenv

Robatronic, 2021-08-12:
@thomascobb , even so, what is meant by "Publishable" at DLS...? :)

Robatronic, 2021-08-12:
@Araneidae , yep, that's where I saw it. :)

thomascobb, 2021-08-13:
A "Publishable IOC" is everything that is needed to run the IOC in a directory on our NFS mounted filesystem. For pythonSoftIOCs, this means both the source files and a Python executable (or link to one) with the correct packages installed.

For DLS we use pipenv to manage the virtual environment of projects, so pythonSoftIOC will be installed with its own pipenv, so the "Single File IOC" can just use that. For more complicated projects (and ones with 3rd party Python module dependencies), we make a Python module, which will have its own pipenv, containing both the Python executable and the packages that module needs

closed 2021-08-31 by thomascobb