DiamondLightSource/pythonSoftIOC #29: An error

Byeongdulee, 2021-08-19:
Hi,
I am trying to make my own motorRecords by modifying db and found examples under tests.
My softioc is pip installed, and it produces the error below:

pythonSoftIOC/tests] python sim_asyncio_ioc_override.py myprefix
Traceback (most recent call last):
  File "sim_asyncio_ioc_override.py", line 36, in <module>
    softioc.iocInit(asyncio_dispatcher.AsyncioDispatcher(event_loop))
TypeError: __init__() takes 1 positional argument but 2 were given

AlexanderWells-diamond, 2021-08-20:
Hi Byeongdulee. I think your problem is here:

> asyncio_dispatcher.AsyncioDispatcher(event_loop)

`AsyncioDispatcher` doesn't accept any arguments. It certainly doesn't want to take an input event loop, as it creates one itself. If you want you can use that created dispatcher's loop for further asyncio processing, something like this:

```
dispatcher = asyncio_dispatcher.AsyncioDispatcher()
asyncio.run_coroutine_threadsafe(my_method(), dispatcher.loop)
```

thomascobb, 2021-08-20:
> ```softioc.iocInit(asyncio_dispatcher.AsyncioDispatcher(event_loop))```

These changes were added in master, but were not released. I will make a release later today and comment on here.

> make my own motorRecords

This will not work at the moment, as pythonSoftIOC does not build in support for the motor record. There are two things that would need to be done to make that work:
- Sort out generic packaging of epics modules as per #6 
- Write Python device support for the motor record

Unfortunately we don't have the time to do either of those at present

Can I ask what is your use case for making an IOC with python device support for a motor record?

Byeongdulee, 2021-08-20:
Thank you Alexander and Tom.
I have been working with a PI piezo stage. The PI has updated its firmware, but the epics driver is not yet as far as I know. Hope to use the DIO of the controller programmatically, for example to send or receive a detector trigger. It could be done in other ways, but since PI's python support is available, I thought pythonsoftioc would be ideal if possible.
Another example is to make a simple IOC for the UR robot.

thomascobb, 2021-08-20:
If you avoid the motor record, then you could do this with pythonSoftIOC. If you need the motor record I'm afraid it won't be suitable at the moment

closed 2021-08-31 by thomascobb