DiamondLightSource/pythonSoftIOC #37: Waveform records ignore NELM keyword if initial_value is set

AlexanderWells-diamond, 2021-09-14:
If you specify both an `initial_value=...` and `NELM=...` keywords when creating a WaveformOut record, the `NELM` keyword is ignored and the waveform is created with a maximum length equal to the length of the data. 

I think the offending line is L169 of builder.py:
`fields['NELM'] = length`

There is a workaround: instead of `initial_value` and `NELM`, use something like this (assuming `data` is a numpy array)
```
length=required_length,
datatype=data.dtype,
```

And later do `myrecord.set(data)`

Araneidae, 2021-09-14:
There are I think two or maybe three issues here:

1. The waveform initialisation API is a little bit too tricksy, and the documentation https://github.com/dls-controls/pythonSoftIOC/blob/ea1b85a3147b673cb864a424b7c59115d9cff7fe/docs/reference/api.rst#L299-L308 feels a little thin on this point.
2. The `NELM` field is being silently overwritten.  This is a clear bug.
3. Perhaps it should be possible to override the length specified in the `initial_value` with a new `length` parameter (subject, I would think to the requirement that the new length be large enough for the value).

I guess the next step is to work out what the initialisation API should be, preferably without breaking reasonable existing usage.

AlexanderWells-diamond, 2021-09-14:
Regarding (1), is there a reason that `value`, `length` and `datatype` keyword arguments exist, while the EPICS keywords `initial_value`, `NELM` and `FTVL` also exist and could be specified on record creation? I can sort of see the reason for datatype, to allow specifying numpy dtypes and have pythonSoftIOC convert it for you, but having two different ways of giving the value and the maximum length seems to just invite issues. It would be nice to know why these exist.

Although at this stage I suppose we have to keep `value` and `length`, as they're in the documentation and so we won't want to break compatibility. 

Regarding (2), I think the fix is simply:
```
# If NELM has been explicitly set use that, otherwise use computed length
if 'NELM' not in fields:
    fields['NELM'] = length
elif fields['NELM'] < length:
    raise ValueError("NELM value shorter than value length")
```

Regarding (3), that's my exact use-case right now. I need to create a very long but mostly empty Waveform, using a short `initial_value` array and a much larger `FTVL` value. I can accomplish this with the above code snippet and providing `initial_value` and `NELM` when I create the record. I cannot currently do it using the `value` and `length` keywords. Should I implement a fix that allows both pairs? Or what about mixing the pairs i.e. `initial_value` and `length`, or `value` and `NELM`?

closed 2022-02-07 by AlexanderWells-diamond