DiamondLightSource/pythonSoftIOC #40: Unable to set StringOut record values

AlexanderWells-diamond, 2021-09-30:
I seem to be unable to set the value of a StringOut record programmatically by calling `.set()`. This program demonstrates the issue:

```
# Import the basic framework components.
from softioc import softioc, builder, asyncio_dispatcher

# Create an asyncio dispatcher, the event loop is now running
dispatcher = asyncio_dispatcher.AsyncioDispatcher()

# Set the record prefix
builder.SetDeviceName("MY-DEVICE-PREFIX")

# Create some records
strout = builder.stringOut("FLIBBLE", initial_value="ABC")

# Boilerplate get the IOC started
builder.LoadDatabase()
softioc.iocInit(dispatcher)

strout.set("DEF")

# Finally leave the IOC running with an interactive shell.
softioc.interactive_ioc(globals())
```

The `strout.set()` line generates this exception:

```
 File "/home/eyh46967/dev/pythonSoftIoc/docs/examples/trial_stringout.py", line 17, in <module>
    strout.set("DEF")
  File "/home/eyh46967/dev/pythonSoftIoc/softioc/device.py", line 212, in set
    datatype, length, data, array = self.value_to_dbr(value)
  File "/home/eyh46967/dev/pythonSoftIoc/softioc/device.py", line 186, in value_to_dbr
    dbrtype = self.NumpyCharCodeToDbr[value.dtype.char]
KeyError: 'U'
```

The `device.py` code appears to be trying to convert the string `value` into a numpy array on line 173:
```
value = numpy.require(value, requirements = 'C')
```

Examining this array gives a `value.dtype.char` of `U`, which is unknown to the `NumpyCharCodeToDbr` mapping dictionary.

Araneidae, 2021-09-30:
Ooh, another interesting one.  This is closely related to #39 and will have to be solved together with that.

AlexanderWells-diamond, 2021-11-18:
We need to create a single function to sanitise all inputs from all possible places (initialising initial values for in & out, setting values for in&out, and waveforms) which can handle all possible Python values and convert it to a valid EPICS value.

Ensure the validation happens before iocInit(), so it'll throw an error if something goes wrong. 

Whenever we pass a value we want to sanitise it to its canonical form - numpy aray for waveforms, or python string, or underlying value for numeric types, and then return the canonicalised form. 

This work will involve device.py, getting and setting the values in some consistent way.