DiamondLightSource/pythonSoftIOC #53: Be able to force Out records into an error state

AlexanderWells-diamond, 2021-10-13:
Currently there is no explicit way to force an Out record into an error state. The `set()` method on In records provides the `severity` and `alarm` keyword arguments (as well as the `set_alarm()` method), but there is no equivalent mechanism for Out records. Providing this would be useful for the PandABlocks-client PythonSoftIOC integration, as PandA can report fields as being in_error, and the associated records are often Out types.

evalott100, 2024-04-04:
@AlexanderWells-diamond

Was there any progress on this? We need it for [this](https://github.com/PandABlocks/PandABlocks-ioc/issues/107).

AlexanderWells-diamond, 2024-04-04:
@Araneidae and I have just discussed this. In principle we can do it. 

There are some behavioural questions though, such as whether this alarm should be "sticky" or whether it should clear on next write, that sort of thing. Could you give us an example of what you'd want/expect to see, assuming we invent some `.set_alarm()` method on out records?

Araneidae, 2024-04-04:
It's worth pointing out that doing this requires a hack and is not directly supported by EPICS.

evalott100, 2024-04-04:
The use case we're looking at is when the PandABlocks ioc stops receiving data from the client. We don't want to end the process, but mirror what other iocs do - keep the PVs active and in an error state.

[image]

`$ caget PANDA1:CALC2:INPA.STAT`
> `PANDA1:CALC2:INPA.STAT         NO_ALARM`

`$ caget PANDA1:CALC2:OUT.STAT`
> `PANDA1:CALC2:OUT.STAT          READ_ACCESS`

PR #157 (2024-04-09): Enable setting alarm status of Out records
Adds the `set_alarm` method to `ProcessDeviceSupportOut`, along with extra keyword arguments to `set`. As these are keyword args with defaults, there should be no compatibility problems.

Tests added to check calling `set_alarm()` before or after IocInit() causes the severity and status attributes to be set as expected.

Closes #53.
closed 2024-04-09 by AlexanderWells-diamond