DiamondLightSource/pythonSoftIOC #68: Some Clarifications Regarding aIn records

Robatronic, 2021-12-11:
It seems there's a variety of ways to define ioc records and I'm looking to confirm or clarify my understanding thus far.

Two main categories are ones using the special PythonIOC device and the lower level ones in builder.records.

I defined:

ai2 = builder.records.ai('AI2', VAL="5", INP="float1 CP MS", SCAN="Passive", PINI="YES")

and updates were seen with a camonitor running in another console.

Conversely:

ai1 = builder.aIn('AI', initial_value=5, INP="float1 CP MS", SCAN="Passive", PINI="YES")

did not update (verified using both caget and camonitor from another console)

So, I set up a monitor on float1 (as per the tutorials) and wrote the value from the callback function. Updates still did not go out. I then changed the SCAN to "I/O Intr" for ai2 and updates started going out.

Do all these "higher-level" records require SCAN="I/O Intr"?
Do they also require that the value is  manually set in Python (i.e. ai.set(value))......?

Or should I be updated ai1 via an ao1's on_update handler....? Is there a "best practice" for what I'm attempting?

Thanks so much for this intriguing tool/utility!!! It holds a lot of promise for us here.  :)

Robatronic, 2021-12-11:
Is PINI meaningful for PVs in this category...?

Sorry for all the questions. Trying to get my head around some of the (what seems to be) nuances.

AlexanderWells-diamond, 2021-12-13:
I should probably leave a full explanation to @Araneidae or @thomascobb , but I can give direct answers based on current code to some of your questions. I'll leave it for others to say if this behaviour is correct and/or could be improved.

By default all `builder.*In` records (i.e. every In record type done through the top level API) are created with `SCAN='I/O Intr'`. (See `_in_record(...)` in  `builder.py`).
Immediately following this, if an `initial_value` is specified `PINI` will be set to `YES`.

It sounds like you may have encountered issue #67 , or some variant thereof, if you are seeing records not update in some cases.

If it's helpful, you can print out the record database file using `builder.WriteRecords("my_records.db")`. Examining the resultant file may help you understand what pythonSoftIOC is doing automatically for you.