DiamondLightSource/pythonSoftIOC #75: Naming convention inconsistency

AlexanderWells-diamond, 2022-01-10:
There's a small naming convention inconsistency with the names of the record types: Action, WaveformIn and WaveformOut are all PascalCase, and every other record is camelCase (aOut, mbbIn, etc.).

On a related note, WaveformIn is defined slightly differently to WaveformOut - it's actually "Waveform" - which causes __name__ to show an unexpected string, and IDE intellisense to highlight it differently

This naming convention has been around for some time, and so changing it would be potentially disruptive.

thomascobb, 2022-01-10:
> On a related note, WaveformIn is defined slightly differently to WaveformOut - it's actually "Waveform" - which causes name to show an unexpected string, and IDE intellisense to highlight it differently

We could swap the alias `WaveformIn` with the definition `Waveform` without breaking anything. We could also change the names and provide aliases to the old ones, but I'm curious as to whether the different names are deliberate.

@Araneidae was there a reason for choosing `builder.WaveformOut` over `builder.waveformOut` like `builder.longOut`?

Personally I prefer to follow PEP8 which would be `builder.waveform_out` but that's probably a step too far

Araneidae, 2022-01-11:
The reason I used `Waveform` rather than `waveform` was to avoid confusion with the `records.waveform` constructor, and I used camelCase for all the others for historical reasons.

I'd rather not change the names.

Araneidae, 2022-01-13:
Don't think this is worth worrying about!

closed 2022-01-13 by Araneidae