DiamondLightSource/pythonSoftIOC #89: Discussion - reccaster like module

tynanford, 2022-05-20:
Hi,

We have a few developers looking into using this python IOC framework and it looks really nice. All our IOCs here include IOC stats and reccaster modules and we would like to keep that going. Looking at the docs, I see devIocStats is already available.

Looks like it would be possible to hook reccaster in a similar way or have reccaster be a separate pip package. From the issue below I see there are issues with that approach. We could also write python code to do what reccaster does or even push PVs directly to CF.

https://github.com/dls-controls/pythonSoftIOC/issues/6

Just wondering if anyone is using channel finder/recsync with these IOCs and what the current thoughts are on EPICS module integration.

Thanks!

tynanford, 2022-05-23:
It is a fairly simple protocol so I don't think it would be too much of an issue to write it in python. I like that approach too, we can look into doing that at ALS. It would mean maintaining both the EPICS module and the python implementations, but in general reccaster doesn't change often so I think that is okay.

coretl, 2022-05-23:
Do you think this should live in another module or in pythonSoftIOC? All code in pythonSoftIOC at the moment works with both cothread and asyncio, which isn't a big deal at the moment as we don't do any socket calls. If we were to add this to pythonSoftIOC then we'd need to make the code support both (cothread uses socket.socket calls, asyncio uses StreamReader and StreamWriter). If it were in another module then you could just support the one you're using...

tynanford, 2022-05-23:
Thanks for pointing out the asyncio vs cothread support, that is good to have in mind going forward. For the moment I'd say we can start with reccaster in a separate module and see how that works.

coretl, 2022-05-23:
Sounds like a good idea. If you ever need to support multiple sorts of I/O in one module, I recommend the [Sans-IO](https://sans-io.readthedocs.io/) approach. An example I have used it for is [PandABlocks-client](https://pandablocks.github.io/PandABlocks-client/master/explanations/sans-io.html)

tynanford, 2022-05-23:
Thanks! We will take a look and keep you updated. Probably won't be able to get to it this week or next

coretl, 2022-05-23:
No problem, I'll leave this issue open for a bit. Would you mind putting a github or pypi link to what you're working on when you're ready then close the issue?

tynanford, 2022-05-23:
Sure, I'll add the github link here once we get started on it.