DiamondLightSource/pythonSoftIOC #90: Install Failed on Windows

Robatronic, 2022-05-27:
I'm not sure what exactly happened, but I've successfully installed previous versions of softIoc. Python version is 3.10.4.

The entirety of the output follows, but I've snipped some highlights first. 

It seemed there were two errors. One earlier having to do with a building a wheel:

"Building wheels for collected packages: softioc
  Building wheel for softioc (pyproject.toml) ... error
  error: subprocess-exited-with-error
  × Building wheel for softioc (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [103 lines of output]"
  
  then later on:
  
  "packages\setuptools_dso\dsocmd.py", line 88, in expand_sources
          raise RuntimeError("Missing source file: %s"%src)
      RuntimeError: Missing source file: iocStats\devIocStats\os\default\osdCpuUsage.c
      [end of output]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for softioc
Failed to build softioc
ERROR: Could not build wheels for softioc, which is required to install pyproject.toml-based projects"

I suspect the latter has to do with the former.

pip install softioc
Collecting softioc
  Using cached softioc-4.0.1.tar.gz (58 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy
  Using cached numpy-1.22.4-cp310-cp310-win_amd64.whl (14.7 MB)
Collecting epicscorelibs<7.0.6.99.3,>=7.0.6.99.2.0
  Using cached epicscorelibs-7.0.6.99.2.0-cp310-cp310-win_amd64.whl (2.6 MB)
Collecting epicsdbbuilder>=1.4
  Using cached epicsdbbuilder-1.5-py3-none-any.whl (23 kB)
Requirement already satisfied: setuptools in c:\users\patelk\appdata\local\programs\python\python310\lib\site-packages (from epicscorelibs<7.0.6.99.3,>=7.0.6.99.2.0->softioc) (58.1.0)
Collecting setuptools-dso>=2.0a1
  Using cached setuptools_dso-2.5-py2.py3-none-any.whl (20 kB)
Building wheels for collected packages: softioc
  Building wheel for softioc (pyproject.toml) ... error
  error: subprocess-exited-with-error
  × Building wheel for softioc (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [103 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-310
      creating build\lib.win-amd64-cpython-310\softioc
      copying softioc\alarm.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\asyncio_dispatcher.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\builder.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\device.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\device_core.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\fields.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\imports.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\pvlog.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\pythonSoftIoc.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\softioc.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\_version_git.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\__init__.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\__main__.py -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\access.acf -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\device.dbd -> build\lib.win-amd64-cpython-310\softioc
      copying softioc\devIocStats.dbd -> build\lib.win-amd64-cpython-310\softioc
      creating build\lib.win-amd64-cpython-310\softioc\iocStatsDb
[... 60 lines cut ...]
          self.distribution.run_command(command)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\dist.py", line 1229, in run_command
          super().run_command(command)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 986, in run_command
          cmd_obj.run()
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools_dso\dsocmd.py", line 509, in run
          _build_ext.run(self)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\command\build_ext.py", line 79, in run
          _build_ext.run(self)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 339, in run
          self.build_extensions()
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 448, in build_extensions
          self._build_extensions_serial()
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools\_distutils\command\build_ext.py", line 473, in _build_extensions_serial
          self.build_extension(ext)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools_dso\dsocmd.py", line 512, in build_extension
          expand_sources(self, ext.sources)
        File "C:\Users\patelk\AppData\Local\Temp\pip-build-env-dm1jp3ze\overlay\Lib\site-packages\setuptools_dso\dsocmd.py", line 88, in expand_sources
          raise RuntimeError("Missing source file: %s"%src)
      RuntimeError: Missing source file: iocStats\devIocStats\os\default\osdCpuUsage.c
      [end of output]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for softioc
Failed to build softioc
ERROR: Could not build wheels for softioc, which is required to install pyproject.toml-based projects

coretl, 2022-06-06:
Looks like https://github.com/dls-controls/pythonSoftIOC/issues/83 which is fixed, but I don't think we made a release of it... @AlexanderWells-diamond I know we're having issues with CI at the moment, is it in a good enough state to make a 4.0.2 release from master?

AlexanderWells-diamond, 2022-06-06:
I don't think we can release at the moment. The CI failure is that we apparently cannot package IocStats into the wheel, which would be a breaking change if released. I've got [a branch](https://github.com/dls-controls/pythonSoftIOC/tree/ci-fixes) for CI work, but am unsure how to fix the issue at the moment.

closed 2022-06-06 by coretl