DiamondLightSource/pythonSoftIOC #118: Support new EPICS 7 records

AlexanderWells-diamond, 2023-01-20:
As PythonSoftIOC already has support for EPICS 7, through `epicscorelibs`, it seems sensible to add support for the new record types available. 

The full list is available here: https://epics.anl.gov/base/R7-0/7-docs/RecordReference.html
The most useful ones on first glance appear to be:
- [Analog Array Input Record (aai)](https://epics.anl.gov/base/R7-0/7-docs/aaiRecord.html)
- [Analog Array Output Record (aao)](https://epics.anl.gov/base/R7-0/7-docs/aaoRecord.html)
- [64bit Integer Input Record (int64in)](https://epics.anl.gov/base/R7-0/7-docs/int64inRecord.html)
- [64bit Integer Output Record (int64out)](https://epics.anl.gov/base/R7-0/7-docs/int64outRecord.html)
- [Long String Input Record (lsi)](https://epics.anl.gov/base/R7-0/7-docs/lsiRecord.html)
- [Long String Output Record (lso)](https://epics.anl.gov/base/R7-0/7-docs/lsoRecord.html)

There are many other record types available (as well as many from previous EPICS versions that we never supported), so some investigation is required to confirm the exact list of useful records to add.

Araneidae, 2024-05-07:
Regarding other record types, I'd suggest only providing driver support for record types which map directly to a primitive type.  I suspect this only includes `int64` and long strings.

Araneidae, 2024-05-09:
Looking at [lsi](https://epics.anl.gov/base/R7-0/7-docs/lsiRecord.html) and [lso](https://epics.anl.gov/base/R7-0/7-docs/lsoRecord.html) I'm disappointed to see two points not properly covered:
* Unicode support _appears_ to be ruled out by the phrase "The ... record is used to [support] an arbitrary ASCII string."  I think we're going to have to continue with our extension to UTF-8 for these new record types.
* Null termination does not appear to be discussed at all: are these strings null terminated?  I guess they must be as there is no separate length field.

If we do support `lsi`/`lso` I suggest calling them `lsIn`, `lsOut` and make them behave just like `stringIn`/`stringOut`.

AlexanderWells-diamond, 2024-06-07:
After some discussion it was decided we should only implement `int64in/out`, as the `lsi` and `lso` records offer no advantages ontop of our already existing `longStringIn/Out` record types. Other new record types do not directly map to Python types.

PR #161 (2024-06-07): Add int64in/out record support
Add support for the [int64in](https://epics.anl.gov/base/R7-0/6-docs/int64inRecord.html) and [int64out](https://epics.anl.gov/base/R7-0/6-docs/int64outRecord.html) record types. 

There's two TODOs and some test failures that needs discussion: 
- The `DBR_*` types we use come from `epicscorelibs`, however its latest version does not define the (new) `DBR_INT64` and `DBR_UINT64` types. Is it worth trying to get a PR through that module that adds these codes before merging this?
- I don't fully understand the note next to DBF_ULONG as to why it is an int32, rather  [... cut]