DiamondLightSource/pythonSoftIOC #121: Add CI support and wheels for aarch64?

Araneidae, 2023-03-09:
At present the only Linux target architecture we support is x86_64, but with the Raspberry Pi becoming more used it may be worth adding CI support for at least aarch64 (ARM v8 64-bit).  This is not altogether straightforward, as this is not natively supported by github, as discussed in https://github.com/actions/runner-images/issues/5631, but it seems that QEMU _can_ be used to work around this, see https://github.com/uraimo/run-on-arch-action.

coretl, 2023-03-09:
cibuildwheel might help with this...

AlexanderWells-diamond, 2023-03-10:
I've been working on this in the `arm_ci` branch. I don't think we can do this currently, as the emulation is both very slow and currently shows a segfault in the tests. See [the run here](https://github.com/dls-controls/pythonSoftIOC/actions/runs/4383709808/jobs/7674284315), with the interesting bit being that it took 1737.29s to build (compared to 9.27s on the native architecture) and once it started tests it failed the first two and segfaulted on the third (and took 1089.21s to do so)