DiamondLightSource/pythonSoftIOC #145: Pin pvxs to earlier version

evalott100, 2023-12-14:
When pandablocks-ioc uses softioc version 4.5.0, the following import error occurs:

```
.venv) [twj43146@pc0139 PandABlocks-ioc-fork]$ pytest
filename="modules/database/src/ioc/dbStatic/dbLexRoutines.c" line number=276dbRead opening file pvxsIoc.dbd
ImportError while loading conftest '/scratch/twj43146/Programming/PandABlocks-ioc-fork/tests/conftest.py'.
tests/conftest.py:3: in <module>
    from fixtures.mocked_panda import *
tests/fixtures/mocked_panda.py:38: in <module>
    from softioc import builder
.venv/lib/python3.10/site-packages/softioc/__init__.py:25: in <module>
    dbLoadDatabase('pvxsIoc.dbd', pvxslibs.path.dbd_path, None)
.venv/lib/python3.10/site-packages/softioc/imports.py:45: in expect_success
    assert status == 0, 'Expected success'
E   AssertionError: Expected success
```

This is caused by the [latest release](https://github.com/mdavidsaver/pvxs/releases/tag/1.3.0) of pvxs.

Pinning ` "pvxslibs==1.2.4"` fixes this problem.

coretl, 2023-12-14:
yes

AlexanderWells-diamond, 2023-12-15:
No longer necessary - see #146 .

closed 2023-12-15 by AlexanderWells-diamond