DiamondLightSource/pythonSoftIOC #169: Install issues with gcc-14.1

codedump, 2024-07-23:
Hello,

`pip install` will bug out with the following error for me:
```
   [...]
gcc -fno-strict-overflow -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -fexceptions -fcf-protection -fexceptions -fcf-protection -fexceptions │
-fcf-protection -fPIC -D_GNU_SOURCE -D_DEFAULT_SOURCE -Dlinux -DUNIX -D_GLIBCXX_USE_CXX11_ABI=1 -I/tmp/pip-build-env-03ykrg0g/overlay/lib64/python3.12/site-pa│
ckages/epicscorelibs/include -Isoftioc/iocStats/devIocStats -Isoftioc/iocStats/devIocStats/os/Linux -Isoftioc/iocStats/devIocStats/os/default -I/usr/include/p│
ython3.12 -c softioc/iocStats/devIocStats/devIocStatsSub.c -o build/temp.linux-x86_64-cpython-312/softioc/iocStats/devIocStats/devIocStatsSub.o -g0 -std=c99  │
      softioc/iocStats/devIocStats/devIocStatsSub.c: In function ‘scanMonInit’:                                                                               │
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: error: implicit declaration of function ‘malloc’ [-Wimplicit-function-declaration]                 │
         91 |   psub->dpvt = malloc(sizeof(epicsTimeStamp));                                                                                                  │
            |                ^~~~~~                                                                                                                           │
      softioc/iocStats/devIocStats/devIocStatsSub.c:26:1: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’                                     │
         25 | #include "devIocStats.h"                                                                                                                        │
        +++ |+#include <stdlib.h>                                                                                                                             │
         26 |                                                                                                                                                 │
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: warning: incompatible implicit declaration of built-in function ‘malloc’ [-Wbuiltin-declaration-mis│
match]                                                                                                                                                        │
         91 |   psub->dpvt = malloc(sizeof(epicsTimeStamp));                                                                                                  │
            |                ^~~~~~                                                                                                                           │
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’                                    │
      error: command '/usr/bin/gcc' failed with exit code 1
```

Essentially, `gcc` is right: there's no `<stdlib.h>` in `devIocStatsSub.c`. I'm guessing it's a 14 year old upstream error of EPICS.

I don't know why it used to work until now, but it might have something to do with the fact that it's a gcc 14.1 environment?... Maybe something changed in the way the standard prototype is being interpreted (which I belive is `int malloc()`, with `()` generally meaning "no arguments", but also being a valid replacement for "(...)" in the past -- maybe not anymore?)

Or it's because of Python 3.12.4, the one I'm using?

I'm running this on what most likely amounts to a clean install of Fedora 40. (Technically, it's a freshly created "Toolbox" on a recent "Fedora CoreOS"...) Here is the complete script to reproduce:
```
toolbox create
toolbox enter
sudo dnf install python3-pip python3-devel
sudo dnf group install "C Development Tools and Libraries"
pip install softioc
```

Cheers,
F.

codedump, 2024-07-23:
I can confirm that it works as intended with an overall older OS version (e.g. `toolbox create --release 39`, which will recreate an environment representing a past Fedora version).

It really is something about the new set of tools.

codedump, 2024-07-23:
Ah. Got it: [Handling of implicit function declarations, new in GCC-14](https://gcc.gnu.org/gcc-14/porting_to.html):

> It is no longer possible to call a function that has not been declared. [...]

I guess this means either a lot of code fixing, or sticking to GCC <= 13 for another decade or so... :facepalm:

Araneidae, 2024-07-23:
This is a bit of a pain because devIocStats is an external dependency, so fixing this is not as simple as fixing our code.  It's probably worth raising an issue with [devIocStats](https://github.com/epics-modules/iocStats/issues) and cross-linking here.  Presumably you can check that this is the only error by adding the appropriate `#include` to your local checkout and checking that building then works.

I suppose we could do some kind of patch to the checked out dependency before building it?

The preferred fix is probably to implement #106.