DiamondLightSource/pythonSoftIOC #171: Dependency Issue with epicscorelibs

XavSPM, 2024-08-21:
Hello,

I attempted to install the latest version of pythonSoftIOC on a Raspberry Pi, but I'm encountering a dependency error with epicscorelibs. I need to use the latest version because I require the `non_interactive_ioc()` function, which is not available in version 4.5.0 on the pip repository.

```bash
epics@Ambiance-P1:~ $ python -m venv test
epics@Ambiance-P1:~/test $ source bin/activate
(test) epics@Ambiance-P1:~ $ python -V
Python 3.11.2
(test) epics@Ambiance-P1:~ $ pip install git+https://github.com/DiamondLightSource/pythonSoftIOC
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/DiamondLightSource/pythonSoftIOC
  Cloning https://github.com/DiamondLightSource/pythonSoftIOC to /tmp/pip-req-build-_y8wk1jc
  Running command git clone --filter=blob:none --quiet https://github.com/DiamondLightSource/pythonSoftIOC /tmp/pip-req-build-_y8wk1jc
  Resolved https://github.com/DiamondLightSource/pythonSoftIOC to commit c110d2fd4616feb4b05377934e37e25a70a0b72d
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting epicscorelibs<7.0.7.99.1,>=7.0.7.99.0.2
  Using cached epicscorelibs-7.0.7.99.0.2-cp311-cp311-linux_aarch64.whl
Collecting pvxslibs>=1.2.2
  Downloading pvxslibs-1.3.1.tar.gz (660 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 660.3/660.3 kB 5.5 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy
  Using cached numpy-2.1.0-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.0 MB)
Collecting epicsdbbuilder>=1.4
  Downloading https://www.piwheels.org/simple/epicsdbbuilder/epicsdbbuilder-1.5-py3-none-any.whl (19 kB)
Requirement already satisfied: setuptools in ./test/lib/python3.11/site-packages (from epicscorelibs<7.0.7.99.1,>=7.0.7.99.0.2->softioc==4.5.0+29.gc110d2f) (66.1.1)
Collecting setuptools-dso>=2.9a1
  Using cached https://www.piwheels.org/simple/setuptools-dso/setuptools_dso-2.11-py2.py3-none-any.whl (23 kB)
Collecting pvxslibs>=1.2.2
  Downloading pvxslibs-1.2.4.tar.gz (643 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 643.9/643.9 kB 12.3 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  Downloading pvxslibs-1.2.3.tar.gz (642 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 642.5/642.5 kB 12.4 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  Downloading pvxslibs-1.2.2.tar.gz (641 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 641.0/641.0 kB 12.6 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
INFO: pip is looking at multiple versions of epicsdbbuilder to determine which version is compatible with other requirements. This could take a while.
Collecting epicsdbbuilder>=1.4
  Downloading https://www.piwheels.org/simple/epicsdbbuilder/epicsdbbuilder-1.4.3-py3-none-any.whl (22 kB)
INFO: pip is looking at multiple versions of epicscorelibs to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of <Python from Requires-Python> to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of softioc to determine which version is compatible with other requirements. This could take a while.
ERROR: Cannot install softioc and softioc==4.5.0+29.gc110d2f because these package versions have conflicting dependencies.

The conflict is caused by:
    softioc 4.5.0+29.gc110d2f depends on epicscorelibs<7.0.7.99.1 and >=7.0.7.99.0.2
    pvxslibs 1.3.1 depends on epicscorelibs<7.0.7.99.2 and >=7.0.7.99.1.1a2
    softioc 4.5.0+29.gc110d2f depends on epicscorelibs<7.0.7.99.1 and >=7.0.7.99.0.2
    pvxslibs 1.2.4 depends on epicscorelibs<7.0.7.99.2 and >=7.0.7.99.1.1a2
    softioc 4.5.0+29.gc110d2f depends on epicscorelibs<7.0.7.99.1 and >=7.0.7.99.0.2
    pvxslibs 1.2.3 depends on epicscorelibs<7.0.7.99.2 and >=7.0.7.99.1.1a2
    softioc 4.5.0+29.gc110d2f depends on epicscorelibs<7.0.7.99.1 and >=7.0.7.99.0.2
    pvxslibs 1.2.2 depends on epicscorelibs<7.0.7.99.2 and >=7.0.7.99.1.1a2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

It seems there's a dependency conflict between epicscorelibs and pvxslibs, which is preventing the installation.

Do you have any suggestions on how to resolve this issue?

AlexanderWells-diamond, 2024-08-21:
Apologies, the current state of main is somewhat broken due to a series of dependencies that are part-way through being updated to support Numpy 2.0, as that requires us to update all of them to build against the newest Numpy version. 

It is currently being tracked in https://github.com/DiamondLightSource/pythonSoftIOC/pull/158

XavSPM, 2024-08-21:
Perfect. I will wait for this update.

I have a few Raspberry Pi running `pythonSoftIOC` on-site for non-critical measurements, and it’s working well so far.

In the future, I also plan to test running `pythonSoftIOC` with the [`revpimodio`](https://revpimodio.org/en/homepage/) library on [Revolution Pi](https://revolutionpi.com/en) devices.

Thank you,

AlexanderWells-diamond, 2024-09-18:
Just to let you know the master branch should now work for you as it supports Numpy2.0, using alpha releases of epicscorelibs, p4p, and pvxslibs.

XavSPM, 2024-09-19:
Hi @AlexanderWells-diamond, 

I've run a test, and can confirm that the problem has been fixed. 

Thanks +1: 

```bash
epics@ambiance-cc:~ $ python -m venv test_pythonsoftIOC
epics@ambiance-cc:~ $ source test_pythonsoftIOC/bin/activate
(test_pythonsoftIOC) epics@ambiance-cc:~ $ python -V
Python 3.11.2
(test_pythonsoftIOC) epics@ambiance-cc:~ $ pip install git+https://github.com/DiamondLightSource/pythonSoftIOC
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/DiamondLightSource/pythonSoftIOC
  Cloning https://github.com/DiamondLightSource/pythonSoftIOC to /tmp/pip-req-build-b12n2y3c
  Running command git clone --filter=blob:none --quiet https://github.com/DiamondLightSource/pythonSoftIOC /tmp/pip-req-build-b12n2y3c
  Resolved https://github.com/DiamondLightSource/pythonSoftIOC to commit 35bb833f14b9818f598e860b3d8d4171f06b7769
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1a3
  Using cached epicscorelibs-7.0.7.99.1.1a3-cp311-cp311-linux_aarch64.whl
Collecting pvxslibs>=1.3.2a2
  Downloading pvxslibs-1.3.2a2.tar.gz (660 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 660.8/660.8 kB 2.4 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy
  Using cached numpy-2.1.1-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.0 MB)
Collecting epicsdbbuilder>=1.4
  Downloading https://www.piwheels.org/simple/epicsdbbuilder/epicsdbbuilder-1.5-py3-none-any.whl (19 kB)
Collecting pyyaml>=6.0
  Downloading PyYAML-6.0.2-cp311-cp311-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (736 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 736.8/736.8 kB 9.0 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./test_pythonsoftIOC/lib/python3.11/site-packages (from epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1a3->softioc==4.5.0+42.g35bb833) (66.1.1)
Collecting setuptools-dso>=2.11a2
  Using cached https://www.piwheels.org/simple/setuptools-dso/setuptools_dso-2.11-py2.py3-none-any.whl (23 kB)
Building wheels for collected packages: softioc, pvxslibs
  Building wheel for softioc (pyproject.toml) ... done
  Created wheel for softioc: filename=softioc-4.5.0+42.g35bb833-cp311-cp311-linux_aarch64.whl size=119544 sha256=44764969b576f792ad04164cd2d2f9a1c78a159cbd3bbac245985405f086c870
  Stored in directory: /tmp/pip-ephem-wheel-cache-ez5uviz_/wheels/d7/6c/1b/e5ddbf1d8c0928c8d135677aac22381d4d638513623e5e1a7e
  Building wheel for pvxslibs (pyproject.toml) ... done
  Created wheel for pvxslibs: filename=pvxslibs-1.3.2a2-cp311-cp311-linux_aarch64.whl size=2267728 sha256=e5bd23f55d38634a378e6cbe3e4e8f1e0d98fb8a9d10548da0f7c3633ca0be85
  Stored in directory: /home/epics/.cache/pip/wheels/e3/f2/e5/0f180b95deec114d79a22d52c3b37fdaf3260ede9408d2ebd3
Successfully built softioc pvxslibs
Installing collected packages: epicsdbbuilder, setuptools-dso, pyyaml, numpy, epicscorelibs, pvxslibs, softioc
Successfully installed epicscorelibs-7.0.7.99.1.1a3 epicsdbbuilder-1.5 numpy-2.1.1 pvxslibs-1.3.2a2 pyyaml-6.0.2 setuptools-dso-2.11 softioc-4.5.0+42.g35bb833
(test_pythonsoftIOC) epics@ambiance-cc:~ $ 
```