DiamondLightSource/pythonSoftIOC #183: Ubuntu VM on Apple Silicon fails to load or compile devIocStats

prichards-wmko, 2025-01-29:
I have softIOC code developed on other systems that is failing to run on my Apple Silicon Macbook Pro M2 (in a Parallels virtual machine running Ubuntu), and it appears to be due to the devIocStats feature causing a buffer overflow and then core dump.

My machine:

```vm-ubuntu24 13:28 ~ $ uname -a
Linux vm-ubuntu24 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:32:09 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

The easiest way to demonstrate the error; this is on a machine installed last July.  I wasn't running devIocStats on it until now.

```vm-ubuntu24 13:27 ~ $ /usr/local/anaconda/bin/python3
Python 3.12.4 | packaged by Anaconda, Inc. | (main, Jun 18 2024, 15:03:55) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from softioc import softioc
INFO: PVXS QSRV2 is loaded, permitted, and ENABLED.
>>> softioc.devIocStats('test')
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```

I also tried to reproduce the problem on a freshly installed Ubuntu 24.04 machine I just made for a test.  It fails to compile.

```# ./pip install softioc
Collecting softioc
  Downloading softioc-4.5.0.tar.gz (82 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1 (from softioc)
  Using cached epicscorelibs-7.0.7.99.1.1-cp312-cp312-linux_aarch64.whl
Collecting pvxslibs>=1.2.2 (from softioc)
  Downloading pvxslibs-1.3.2.tar.gz (660 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 660.7/660.7 kB 411.8 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /usr/local/Anaconda3-2024.10-1/lib/python3.12/site-packages (from softioc) (1.26.4)
Collecting epicsdbbuilder>=1.4 (from softioc)
  Downloading epicsdbbuilder-1.5-py3-none-any.whl.metadata (2.8 kB)
Requirement already satisfied: setuptools in /usr/local/Anaconda3-2024.10-1/lib/python3.12/site-packages (from epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1->softioc) (75.1.0)
Collecting setuptools-dso>=2.11a2 (from epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1->softioc)
  Using cached setuptools_dso-2.11-py2.py3-none-any.whl.metadata (1.6 kB)
Downloading epicsdbbuilder-1.5-py3-none-any.whl (23 kB)
Using cached setuptools_dso-2.11-py2.py3-none-any.whl (23 kB)
Building wheels for collected packages: softioc, pvxslibs
  Building wheel for softioc (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for softioc (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [149 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/__init__.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/__main__.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/_version_git.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/alarm.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/asyncio_dispatcher.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/builder.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/cothread_dispatcher.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/device.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/device_core.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/fields.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/imports.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/pvlog.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/pythonSoftIoc.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/softioc.py -> build/lib.linux-aarch64-cpython-312/softioc
      copying softioc/access.acf -> build/lib.linux-aarch64-cpython-312/softioc
[... 116 lines cut ...]
      gcc -pthread -B /usr/local/Anaconda3-2024.10-1/compiler_compat -fno-strict-overflow -Wsign-compare -DNDEBUG -O3 -Wall -fPIC -O3 -isystem /usr/local/Anaconda3-2024.10-1/include -fPIC -O3 -isystem /usr/local/Anaconda3-2024.10-1/include -fPIC -D_GNU_SOURCE -D_DEFAULT_SOURCE -Dlinux -DUNIX -D_GLIBCXX_USE_CXX11_ABI=1 -I/tmp/pip-build-env-f4dm2sp3/overlay/lib/python3.12/site-packages/epicscorelibs/include -Isoftioc/iocStats/devIocStats -Isoftioc/iocStats/devIocStats/os/Linux -Isoftioc/iocStats/devIocStats/os/default -I/usr/local/Anaconda3-2024.10-1/include/python3.12 -c softioc/iocStats/devIocStats/devIocStatsSub.c -o build/temp.linux-aarch64-cpython-312/softioc/iocStats/devIocStats/devIocStatsSub.o -g0 -std=c99
      softioc/iocStats/devIocStats/devIocStatsSub.c: In function ‘scanMonInit’:
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: error: implicit declaration of function ‘malloc’ [-Wimplicit-function-declaration]
         91 |   psub->dpvt = malloc(sizeof(epicsTimeStamp));
            |                ^~~~~~
      softioc/iocStats/devIocStats/devIocStatsSub.c:26:1: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’
         25 | #include "devIocStats.h"
        +++ |+#include <stdlib.h>
         26 |
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: warning: incompatible implicit declaration of built-in function ‘malloc’ [-Wbuiltin-declaration-mismatch]
         91 |   psub->dpvt = malloc(sizeof(epicsTimeStamp));
            |                ^~~~~~
      softioc/iocStats/devIocStats/devIocStatsSub.c:91:16: note: include ‘<stdlib.h>’ or provide a declaration of ‘malloc’
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for softioc
  Building wheel for pvxslibs (pyproject.toml) ... done
  Created wheel for pvxslibs: filename=pvxslibs-1.3.2-cp312-cp312-linux_aarch64.whl size=1947218 sha256=4d61814d8b17ea04a967b5889969140bac7584c0299ea5937933f3e18586de4b
  Stored in directory: /root/.cache/pip/wheels/cb/b2/4e/5385924e571775421b6050a485cc1960c26227e972b21026c8
Successfully built pvxslibs
Failed to build softioc
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (softioc)
```

AlexanderWells-diamond, 2025-01-29:
Sorry for the issues you are encountering. We unfortunately have no MacOS machines to test against  - and one of our dependencies also [does not currently support arm64](https://github.com/DiamondLightSource/cothread/issues/68).

If you are willing to test it, I have a potential solution: could you please try installing pythonSoftIOC directly from a branch using this command:
```
pip install git+https://github.com/DiamondLightSource/pythonSoftIOC.git@issue_183
```

Hopefully that will compile and run successfully!

prichards-wmko, 2025-01-29:
This installed as expected but did not fix the underlying issue.

First, the install, if it helps to see it:

```
# /usr/local/anaconda/bin/pip install git+https://github.com/DiamondLightSource/pythonSoftIOC.git@issue_183
DEPRECATION: Loading egg at /usr/local/anaconda3-2024.06/lib/python3.12/site-packages/Adafruit_BBIO-1.2.0-py3.12-linux-aarch64.egg is deprecated. pip 24.3 will enforce this behaviour change. A possible replacement is to use pip for package installation.. Discussion can be found at https://github.com/pypa/pip/issues/12330
Collecting git+https://github.com/DiamondLightSource/pythonSoftIOC.git@issue_183
  Cloning https://github.com/DiamondLightSource/pythonSoftIOC.git (to revision issue_183) to /tmp/pip-req-build-c4tzm0j3
  Running command git clone --filter=blob:none --quiet https://github.com/DiamondLightSource/pythonSoftIOC.git /tmp/pip-req-build-c4tzm0j3
  Running command git checkout -b issue_183 --track origin/issue_183
  Switched to a new branch 'issue_183'
  branch 'issue_183' set up to track 'origin/issue_183'.
  Resolved https://github.com/DiamondLightSource/pythonSoftIOC.git to commit 2c7308fa4151f7dd0673439e7f0a14f5418bcc35
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1 (from softioc==4.5.0+53.g2c7308f)
  Using cached epicscorelibs-7.0.7.99.1.1-cp312-cp312-linux_aarch64.whl
Collecting pvxslibs>=1.3.2a2 (from softioc==4.5.0+53.g2c7308f)
  Downloading pvxslibs-1.3.2.tar.gz (660 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 660.7/660.7 kB 2.1 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /usr/local/anaconda3-2024.06/lib/python3.12/site-packages (from softioc==4.5.0+53.g2c7308f) (1.26.4)
Requirement already satisfied: epicsdbbuilder>=1.4 in /usr/local/anaconda3-2024.06/lib/python3.12/site-packages (from softioc==4.5.0+53.g2c7308f) (1.5)
Requirement already satisfied: pyyaml>=6.0 in /usr/local/anaconda3-2024.06/lib/python3.12/site-packages (from softioc==4.5.0+53.g2c7308f) (6.0.1)
Requirement already satisfied: setuptools in /usr/local/anaconda3-2024.06/lib/python3.12/site-packages (from epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1->softioc==4.5.0+53.g2c7308f) (69.5.1)
Collecting setuptools-dso>=2.11a2 (from epicscorelibs<7.0.7.99.2,>=7.0.7.99.1.1->softioc==4.5.0+53.g2c7308f)
  Using cached setuptools_dso-2.11-py2.py3-none-any.whl.metadata (1.6 kB)
Using cached setuptools_dso-2.11-py2.py3-none-any.whl (23 kB)
Building wheels for collected packages: softioc, pvxslibs
  Building wheel for softioc (pyproject.toml) ... done
  Created wheel for softioc: filename=softioc-4.5.0+53.g2c7308f-cp312-cp312-linux_aarch64.whl size=120127 sha256=1607d0a17d35f3de38f5dd97e1c9f50186c4cd246773d541cdfdbf3a5c29d9d9
  Stored in directory: /tmp/pip-ephem-wheel-cache-d0xd9uz8/wheels/10/16/bd/40b31b942f84a986e5010316fe2c4382fa39cb10998831feaa
  Building wheel for pvxslibs (pyproject.toml) ... done
  Created wheel for pvxslibs: filename=pvxslibs-1.3.2-cp312-cp312-linux_aarch64.whl size=1950003 sha256=6411f1de6d2796090a34678f6d2920afd66859b4b1497402a042bdf4fdc0dd6f
  Stored in directory: /root/.cache/pip/wheels/cb/b2/4e/5385924e571775421b6050a485cc1960c26227e972b21026c8
Successfully built softioc pvxslibs
Installing collected packages: setuptools-dso, epicscorelibs, pvxslibs, softioc
  Attempting uninstall: setuptools-dso
    Found existing installation: setuptools-dso 2.10
    Uninstalling setuptools-dso-2.10:
      Successfully uninstalled setuptools-dso-2.10
  Attempting uninstall: epicscorelibs
    Found existing installation: epicscorelibs 7.0.7.99.0.2
    Uninstalling epicscorelibs-7.0.7.99.0.2:
      Successfully uninstalled epicscorelibs-7.0.7.99.0.2
  Attempting uninstall: pvxslibs
    Found existing installation: pvxslibs 1.3.1
    Uninstalling pvxslibs-1.3.1:
      Successfully uninstalled pvxslibs-1.3.1
  Attempting uninstall: softioc
    Found existing installation: softioc 4.5.0
    Uninstalling softioc-4.5.0:
      Successfully uninstalled softioc-4.5.0
Successfully installed epicscorelibs-7.0.7.99.1.1 pvxslibs-1.3.2 setuptools-dso-2.11 softioc-4.5.0+53.g2c7308f

```

The epicscorelibs packages it's own libstdc++.so.6 version for some reason.  This does not match the one in my Ubuntu release and it causes the module to fail to load:

```
$ /usr/local/anaconda/bin/python3
Python 3.12.4 | packaged by Anaconda, Inc. | (main, Jun 18 2024, 15:03:55) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from softioc import softioc
Traceback (most recent call last):
[... 1 lines cut ...]
  File "/usr/local/anaconda/lib/python3.12/site-packages/softioc/__init__.py", line 8, in <module>
    from epicscorelibs.ioc import \
  File "/usr/local/anaconda/lib/python3.12/site-packages/epicscorelibs/ioc.py", line 22, in <module>
    pvAccessIOC = ctypes.CDLL(path.get_lib("pvAccessIOC"), mode=ctypes.RTLD_GLOBAL)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/anaconda/lib/python3.12/ctypes/__init__.py", line 379, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
$ /usr/local/anaconda/bin/python3
Python 3.12.4 | packaged by Anaconda, Inc. | (main, Jun 18 2024, 15:03:55) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from softioc import softioc
INFO: PVXS QSRV2 is loaded, permitted, and ENABLED.
>>> softioc.devIocStats('test')
*** buffer overflow detected ***: terminated
Aborted (core dumped)
```

I'm willing to assist on this further.  It's not critical that I make it work in this VM but it makes some of my testing easier.  Thanks!

AlexanderWells-diamond, 2025-01-30:
Thank you for the information. I believe this issue to be a replica of https://github.com/epics-base/epicscorelibs/issues/32 - it seems `epicscorelibs` doesn't currently play nicely with the latest GCC compilers and their default macros.

Until there is a proper release, the workaround for you is to install `epicscorelibs` from its master branch:

```
pip install git+https://github.com/epics-base/epicscorelibs.git softioc
```

Please let me know if that works, if so we can close out this issue. Thanks.

AlexanderWells-diamond, 2025-02-05:
`epicscorelibs` has now released version `7.0.7.99.1.2a1`, which in my testing appears to fix the issue. I'm closing this for now; please feel free to re-open it if the problem persists.

closed 2025-02-05 by AlexanderWells-diamond

prichards-wmko, 2025-02-07:
I reinstalled softIOC with the latest ``episcorelibs`` and tested it successfully against the latest (``pip install --upgrade git+https://github.com/DiamondLightSource/pythonSoftIOC.git``).  Thanks for the fix!

```
vm-ubuntu24 12:40 ~ $ /usr/local/anaconda/bin/python3
Python 3.12.4 | packaged by Anaconda, Inc. | (main, Jun 18 2024, 15:03:55) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from softioc import softioc
INFO: PVXS QSRV2 is loaded, permitted, and ENABLED.
>>> softioc.devIocStats('test')
>>> 
```