DiamondLightSource/pythonSoftIOC #188: `RuntimeError: Unable to identify compiler cc -pthread` on `ubuntu18.04`, `uv` virtualenv `3.11.9`

konosubakonoakua, 2025-04-21:
## issue
`uv pip install softioc`

got:

```
❯ uv pip install softioc pcaspy pydm
Using Python 3.11.11 environment at: .virtualenvs/py311
Resolved 17 packages in 9ms
  × Failed to build `softioc==4.5.0`
  ├─▶ The build backend returned an error
  ╰─▶ Call to `setuptools.build_meta:__legacy__.build_wheel` failed (exit status: 1)

      [stderr]
      Traceback (most recent call last):
        File "<string>", line 14, in <module>
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/setuptools/build_meta.py", line 331, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/setuptools/build_meta.py", line 301, in _get_build_requires
          self.run_setup()
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/setuptools/build_meta.py", line 512, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/setuptools/build_meta.py", line 317, in run_setup
          exec(code, locals())
        File "<string>", line 5, in <module>
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/epicscorelibs/path/__init__.py", line 5, in <module>
          from ..config import get_config_var
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/epicscorelibs/config.py", line 86, in <module>
          _config = _makeconf()
                    ^^^^^^^^^^^
        File "/home/rf/.cache/uv/builds-v0/.tmpPNDygS/lib/python3.11/site-packages/epicscorelibs/config.py", line 40, in _makeconf
          raise RuntimeError("Unable to identify compiler %s"%CC)
      RuntimeError: Unable to identify compiler cc -pthread

      hint: This usually indicates a problem with the package or the build environment.
```

## workaround
use python 3.10

AlexanderWells-diamond, 2025-04-22:
This appears to be an issue where the wheels for `4.6.0` were not uploaded to PyPi. This is due to a workflow change made between `4.5.0` and `4.6.0`. I'm unsure why it would work for Python3.10 with the same version of `softioc` though, as there's no fundamental difference I'm aware of between the Python versions... 

I'll see about uploading the wheels correctly.

As for the error itself, it implies that the system could not find any of the supported compilers - `gcc`, `clang`, or `msvc`(if on Windows). For some reason it could only find the `cc` compiler.

closed 2025-04-22 by AlexanderWells-diamond

AlexanderWells-diamond, 2025-04-22:
This should now be fixed with the release of `4.6.1`. Thank you for reporting it.

PR #189 (2025-04-22): Fix releasing wheels for all platforms
An after-effect of PR #163 means that artifacts now have unique names per job.
We have to ensure we download all of them to release all wheels.

Fixes #188